c42f/displaz #38: Error while building ilmbase as external project

gadomski, 2015-02-02:
Working from v0.2-171-g0a37088 on OSX 10.10.2, cmake version 3.1.1, I get the following when I run `ninja ilmbase` from my out-of-tree build:

```
[1/3] Performing build step for 'ilmbase'
FAILED: cd /Users/gadomski/Repos/displaz/build/thirdparty/external/ilmbase-prefix/src/ilmbase-build && /usr/local/Cellar/cmake/3.1.1/bin/cmake --build /Users/gadomski/Repos/displaz/build/thirdparty/external/ilmbase-prefix/src/ilmbase-build --config . && /usr/local/Cellar/cmake/3.1.1/bin/cmake -E touch /Users/gadomski/Repos/displaz/build/thirdparty/external/ilmbase-prefix/src/ilmbase-stamp/ilmbase-build
ninja: error: '/Users/gadomski/Repos/displaz/build/thirdparty/external/ilmbase-prefix/src/ilmbase/Half/toFloat.h', needed by 'Half/CMakeFiles/Half.dir/half.cpp.o', missing and no known rule to make it
ninja: build stopped: subcommand failed.
```

There does not seem to be a `src/ilmbase/Half/toFloat.h` in my `ilmbase-prefix` build directory. AFAICT `toFloat.h` hasn't been created by the toFloat custom command in `ilmbase/Half/CMakeLists.txt`, but I can't quite suss out why.

c42f, 2015-02-03:
Yeah, this is a weird one.  `toFloat.h` is autogenerated by the ilmbase build system which is why odd things happen with it occasionally, but I've not been able to reproduce the problem under controlled circumstances.  I'll try upgrading to  ilmbase-2.2.0, that will avoid the need for me to be patching their build system and maybe this problem will go away too.

I've also been thinking about taking the third party build out of the main build, and requiring it to be done manually in a separate step since it seems to be hard to make it really portable.  Aside from that, I've been considering replacing ilmbase with Eigen since it's a bit more standard and will help me to rip out some really nasty casting which I'm currently doing.

gadomski, 2015-02-03:
First, :+1: Eigen, or maybe even [armadillo](http://arma.sourceforge.net/) depending on how heavy-duty you need to get.

Second, I'm building on OSX and I have `brew install ilmbase` available to me. However, I was spooked by the cmake setup into thinking that a system-installed `ilmbase` was sketchily supported or not supported at all. If things *should* work with a system-installed ilmbase, let me know; I'll try again that way and close if that works a'ight.

c42f, 2015-02-03:
The system-supplied ilmbase should probably work.  I have a vague memory that >=ilmbase-2.0.0 was required which was why I built it into the build system, but it looks like I didn't document that.

I'm aware of armadillo but I haven't used it - any thoughts vs Eigen on the speed and API fronts?  It looks like Eigen is more self contained so that probably settles it since my usage of it will be fairly lightweight.

gadomski, 2015-02-03:
Re: armadillo I use it [here](https://github.com/gadomski/cpd/) and I like it a lot. Speed-wise it's all backed up by blas, lapack, arpack, etc, so big operations are pretty good. You can plug in alternatives too (e.g. openblas for blas), and avoid dependencies you don't use (some projects don't use arpack stuff).

For simple matrix operations I can't speak to speed vs. Eigen, but they're both more or less the same category of solution to the linear algebra in C++ problem space — template libraries that let you do fancy stuff. I ended up in armadillo b/c its linear solvers were more robust (IMO) than Eigen's, and I like its API better (maps from Matlab pretty directly, which is nice when porting code written by scientists). But Eigen seems more widely used and if its what you know — a'int broke don't fix?

Re: ilmbase, see #39.

gadomski, 2015-02-05:
Closed by #39.

closed 2015-02-05 by gadomski