c42f/displaz #45: Axis labels in 3D view

c42f, 2015-06-06:
Currently there's no axis labels or coordinate tick labels so it can be hard to orient yourself in the 3D view.  I suppose this could be done by hooking into Qt's font rendering system and sending the resulting textures to OpenGL.

nigels-com, 2015-06-07:
I'd love an excuse to play with NV_path_rendering, ought to be good for 2D annotation and text.  But alas, it's Nvidia only, AFAIK.

https://developer.nvidia.com/nv-path-rendering
http://www.slideshare.net/Mark_Kilgard/an-introduction-to-nvpathrendering

c42f, 2015-06-07:
That looks really cool if only it were portable.  The path rendering API looks disturbingly complex, but I suspect needfully so... they seem to be aiming at postscript-like fill and stroke capability.

nigels-com, 2015-06-07:
It does aim for feature-for-feature compatibility with ps, pdf and flash specifications, so it does have a lot of functionality rolled into it.

Last time I did much with 2D text I think we used OGLFT, but I don't get the impression that's been updated for core OpenGL contexts.  It does have some Qt integration, which was handy for unicode string support.

http://oglft.sourceforge.net/

c42f, 2015-06-07:
Yeah unfortunately the OGLFT svn repo indicates it's not actively maintained since 2008 which doesn't bode well.

nigels-com, 2015-06-07:
An easy kind of thing to do for orientation is to draw red-green-blue axes into the corner of the viewport.
Such as here:  http://www.perl.com/pub/2005/02/17/graphics/step049.png

c42f, 2015-06-07:
Call me picky, but I've always found the RGB axis solution obscure to use and a bit ugly.  I think axis labels would be much nicer.

nigels-com, 2015-06-07:
Ugly and obscure, yes.  But useful, and something that can be toggled via a menu item, if necessary.

nigels-com, 2015-06-08:
So aside from red-green-blue, author some textures for x, y, z labels?
Three white lines, with white x/y/z labels, with baked-in drop-shadows for contrast?

c42f, 2015-06-08:
Yes, something like that was what I had in mind.

Actually I was considering labeling three sides of the data bounding box rather than having a separate set of axes.  This seems to be more or less the standard in scientific plotting programs.  Though... this makes less sense if there's a lot of bounding boxes.  Various text placement options are worth trying I guess - once the infrastructure is in place to get text in there

As to the colour - definitely some kind of shadows/background as necessary for contrast.  Maybe we try making the text background the same colour as the scene background?  Then it's not too ugly, but helps with contrast when necessary.

nigels-com, 2015-06-08:
I put something together on the master branch of my fork:
https://github.com/nigels-com/displaz

The code needs some further tidy-up, and maybe visual tweaking based on your feedback.

c42f, 2015-06-08:
Cool, I'm looking forward to trying it out.  May have to wait until early this afternoon since I'm on child minding duty this morning.

nigels-com, 2015-07-14:
Anything else needed for closing out this issue?
The X/Y/Z vector correspondence?

- Nigel

c42f, 2015-07-14:
Not exactly, I just forgot to close it after closing the PR.

However after using it a bit, there's been some discussion at work that it's still not quite right and I'm inclined to agree: In many situations the perspective axes are hard to interpret in a similar way to the orthographic axes.

I did a little UI prototyping to try clearing up the difficulty - something I came across that might be preferable is to allow people to place the axes on the actual geometry with a mouse click.  That way they're fully in the context of the 3D scene.

As usual, the problem is making this a discoverable part of the interface.  I might end up needing some kind of context-dependent toolbox in a dock window.  Or a toolbar, ugh.

nigels-com, 2015-07-15:
I see some merit in being able to place axes in the world for reference.

For the "orthographic" axes, the way to think of it is that it corresponds to the orientation of the viewer in 3D space, rather than what happens to be the geometry projected at that region of the viewport.  In a similar way to a compass needle.  You wouldn't apply perspective correction to a compass needle either, right?

- Nigel

c42f, 2015-07-15:
I can see your point, but I think both options take some getting used to, and are weird in their own ways.  But that just means neither option is "the right thing".  The design space seems large, surely there's a third option where a random person can sit down and immediately interpret the orientation of the data.

c42f, 2015-11-28:
Well, we've certainly got axes, they seem useful if not perfect, so I'm just going to close this for now.

closed 2015-11-28 by c42f