c42f/displaz #52: Dynamic near clipping plane

c42f, 2015-06-12:
Setting the OpenGL clipping planes are tricky: you want maximum precision from the depth buffer to avoid z fighting, but the geometry has unknown size.  Might be best just to choose it dynamically per frame.

The Outerra blog has a wonderful post about depth buffer precision and analysis of why a perspective correct depth buffer is actually the wrong thing from the point of view of maximizing the available precision: http://outerra.blogspot.com.au/2012/11/maximizing-depth-buffer-range-and.html

gphysics, 2015-06-15:
The far clipping plane distance should probably be set to some multiple of the diagonal length of the bounding box for the geometries selected (I find this box when the Z key is pressed in my axis_alignment branch). The near clipping plane should probably be set to some value that is geometry specific. In the case of point arrays we could work out the rms nearest point separation during the octree sorting process and use this.

c42f, 2015-06-15:
I do want all the geometry which is selected to be visible (ie, not clipped), so the far clipping plane can't be less than the maximum distance from the camera to all selected geometry which lies in the camera frustum.  Using bounding boxes for the geometry would get a quick estimate of the distance.

I agree that near clipping is less clear cut.  Doing something along the lines of estimating the data density probably does make sense.