c42f/displaz #53: Measurement tools

c42f, 2015-06-13:
It's quite useful to be able to measure distances quickly.  The basic idea would be to allow two or more points to be selected as a set of points of interest, and draw lines between these with some graphical representation of the distance (also worth putting it in the log window for copy&paste-ability).  The (a?) 3D cursor is the obvious graphical element for creating user-defined vertices.  Whatever interface we use, the user should be free to move around with the usual controls between placing of the measurement vertices.  3D programs which don't allow camera motion when in "object creation mode" are irritating.  wings3d is an example of a modeling program which correctly allows camera motion even in the middle of creating a geometric primitive.

potree has some useful ideas for this, so it's worth looking at as an example:
http://potree.org/demo/potree_2014.12.30/examples/viewer.html

See also the closely related #26 - these should share almost all the same UI elements; maybe they're even the same feature but with "measurement objects" being temporary by default.

roames, 2016-06-09:
It sounds like this could be easily be build with the help of: #92 (#108) - we could potentially combine that with a scriptable user interface (e.g. https://github.com/wjakob/nanogui) and create all kinds of cool plugins / add-ons. It would also open the door for more contributions.

c42f, 2016-06-09:
Yeah, it'd be great for us hackers :-)

The main question in my mind is how to distribute scripts and have end users able to run them.  Maybe Qt Script would do it for "official" plugins.