c42f/displaz #64: Crash on master, windows 7

c42f, 2015-07-12:
The windows build starts correctly, but segfaults when any data is loaded.

c42f, 2015-07-12:
This seems to be happening as of e6ac5db or so, but it's hard to see how that change could have caused the problem.  Have tried several things without much success:

* Running under valgrind memcheck on linux doesn't reveal anything much
* Windbg says the segfault is in `QAbstractItemView::viewOptions()` but this could just be a side effect of memory corruption coming from somewhere else.  Commenting out anything to do with the data set QListView does seem to allow data to load.

c42f, 2015-07-12:
Hmm, seems to be toolchain dependent.  With a new install of the latest windows SDK 7.1 the crash happens even in version 0.3.1

nigels-com, 2015-07-14:
I've been setting up a Windows 7 laptop, and I was able to build and run with Qt 4.8.6, Visual Studio Express 2013, no crash here.  I can upload the installer, if it's any help.

c42f, 2015-07-14:
Thanks for the offer, though I really need a sustainable solution (fix the bug, or figure out which toolchain+Qt combination to avoid if it's a toolchain issue). I'm most of the way through setting up a new VS2013 install here so I'll see how that goes - certainly quite useful to know that it worked for you, thanks.  All the official releases thus far have been built with VS2010 which is fairly old now, and Qt-4.* is progressively being abandoned.

Actually, it might not be obvious but I've sunk quite a lot of time into this issue so far :-/  I was planning to take a new tack by porting to Qt5 which is actually supported.

c42f, 2015-07-14:
Well you're right, this issue mysteriously goes away by compiling with VS2013.  I'm somewhat inclined to write it off as a library/toolchain incompatibility and quit wasting time on it.  The windows 7.1 SDK is pretty old, so I'm not unhappy to just consider it blacklisted.

nigels-com, 2015-07-14:
With VS2015 landing soon, it seems reasonable to me to require 2013 onwards.

c42f, 2015-07-14:
Yeah, especially since displaz isn't a library which needs to work with other people's arbitrary toolchains.  I'm going to just add a note to the README and good riddance to the old versions.  Thanks for the input.

Semi-related - I created an appveyor account a couple of months ago in an attempt to have some kind of continuous build for windows.  They have qt 5 support so I'll probably give it a proper go once I've tackled that.

c42f, 2015-07-14:
Ok, 30b3167 "addresses" this by warning people against using old toolchains.  I'm now going to start ignoring this and just hope it goes away.  Starting with closing the issue ;-)

closed 2015-07-14 by c42f