c42f/displaz #65: Port to Qt-5

c42f, 2015-07-14:
Qt-4.* is basically out of support, and depending on it is a particular problem on OSX due to OpenGL integration issues.  It would also be good to have on windows to ease the creation of a windows continuous build on appveyor.

c42f, 2015-07-14:
@dedalusj has done some initial work in this direction - see https://github.com/dedalusj/displaz/tree/qt5

dedalusj, 2015-07-15:
Would you like a pull request on a branch for this or you prefer to do the integration manually?

c42f, 2015-07-16:
Not quite sure I understand.  A normal pull request would be great, I'm not sure what else would be better?

chrisidefix, 2015-10-10:
To progress with this, we need to define an acceptable `Qt 5.x` version that people are expected to use. My main concern is related to upgrading from [`QGLWidget`](http://doc.qt.io/qt-5/qglwidget.html) to [`QOpenGLWidget`](http://doc.qt.io/qt-5/qopenglwidget.html), which was only introduced in `Qt 5.4`, while `QGLWidget` has now been tagged as obsolete.

Packages for `Qt 5.4` (or `5.5`) are not yet available on commonly used linux distributions. Therefore, would it be ok that people just have to compile and link their own `Qt` version or would you prefer to be more backwards compatible and keep relying on `QGLWidget` for now?

nigels-com, 2015-10-10:
Looking at the version history, Qt5 has only been around since late 2012.  And Ubuntu 12.04 doesn't include Qt5, except via ppa repositories.  Ubuntu 14.04 ships with Qt 5.2.  And the upcoming 15.10 appears to be 5.4.  So my impression is that 5.4 is too recent to require.  I'd suggest Qt 5.0 as a baseline, providing that compiling with Qt 5.x also works.

My main suggestion is to focus on the core-context problem first, and deferring the Qt 5 migration for a subsequent effort.  Is Qt 4.x still a viable option on a Mac?

- Nigel

chrisidefix, 2015-10-10:
Qt4 is still the standard most projects rely on in my experience. It does seem like a logical step to use the latest QOpenGL* implementation rather than the obsolete QGL* equivalents, though. Technically, there should be a solution for running this on Mac OS X with Qt 4.8.x and I did manage to get a proper OpenGL context setup (by tweaking the Mac OS X window setup specifically), but I am still missing something to get the Framebuffers working with this properly.

c42f, 2015-10-10:
I changed the title since we've mostly been discussing OS X support here.  I'm not set on Qt5 per se - if there's a way to get things working with 4.8 I'd prefer that for the short to medium term since it eases dependencies on linux.

I read the qt blog post about QOpenGLWidget, and although it will make a few things cleaner I'm not sure displaz really needs the new features (https://blog.qt.io/blog/2014/09/10/qt-weekly-19-qopenglwidget).

chrisidefix, 2015-10-11:
**OpenGL version (context) under Mac OS X**

Apparently, you can only access certain OpenGL versions under OS X (including 2.1, 3.2/3.3 or 4.1) as shown here: https://developer.apple.com/opengl/capabilities/index.html

`displaz` appears to rely on a number of OpenGL legacy fixed function pipeline calls. It also required a minimum GLSL shader `#version 130`. This implies an OpenGL version requirement of 3.0, which quite simply won't work for OS X as there is no actual support for the OpenGL compatibility profile under OS X (https://www.opengl.org/wiki/Core_And_Compatibility_in_Contexts).

Therefore, we would either have to move everything to OpenGL 3.2 and replace the fixed function pipeline (using vertex buffers, etc) or we would have to downgrade to OpenGL 2.1 and not use any of the more modern OpenGL functions.

nigels-com, 2015-10-11:
Yes, precisely.

c42f, 2015-10-11:
Yup, see also #7

chrisidefix, 2015-10-11:
Would be good to define which OpenGL version you want to set as a minimum requirement. I saw something about `Intel HD 3000` in the code and I am not sure that supports `OpenGL 3.2` by now?

nigels-com, 2015-10-11:
Something like OpenGL 3.0 core profile + selected extensions, I'd expect.

chrisidefix, 2015-10-11:
There is also an issue with compiling the shaders (through `QGLShaderProgram`) . I am not yet sure how exactly this is related, but the `QGLContext` for `QGLWidget` appears to have some trouble accepting the requested `QGLFormat` under OS X.

When done directly with OpenGL calls (using `glCompileShader`) everything works fine, but both `QGLShaderProgram` and `QGLFramebufferObject` don't seem to play along well under Mac OS X, when you want to use OpenGL > 2.1.

Right now, I would recommend to remove these and replace them with native OpenGL calls, which would also make the transition to Qt 5.x easier later on. It does seem to be a bit of work to replace the `QGL*` code, though and I would appreciate your input on this first.

nigels-com, 2015-10-11:
Yeah, it's a shame to have to go re-implement things for Core context support, but sounds like a reasonable approach, given the circumstances.  It would be nice to be able to target Qt 4.8 or 5.x, for example.  Does that sound feasible?

c42f, 2015-10-11:
Version wise, I've been trying to limit it to version 3.3 or so since that seems to have most of the functionality which is strictly necessary (though far from all that is desirable), and it's fairly compatible with systems out there.

I've no particular love of the QGL 4.x stuff because it doesn't expose all the OpenGL functionality that I might like, hence various additional wrapper code in glutil.h.  It's probably not too hard to wrap the specific parts we need.  Nigel - I suspect you have a better idea about that than me.

Sigh... I was hoping I'd be able to throw out most of the glutil code when porting to Qt5 :-)

nigels-com, 2015-10-12:
As a data-point, if Mesa3D is of interest (for say, offscreen rendering):
`OpenGL version 3.0 Mesa 10.5.9 is supported` (Ubuntu 15.04)

c42f, 2016-01-10:
Some niggling issues no doubt remain, but I think this is done for now.

closed 2016-01-10 by c42f