c42f/displaz #86: Segfault after modifying dataset list

c42f, 2015-12-13:
To reproduce:

* Open two laz files
* Open Data Sets tab
* Select second data set with mouse; press delete to remove it from the list
* Select first data set; displaz will now segfault :-(

According to gdb, the crash is inside the OpenGL driver, as called from `pointarray.cpp:721`.

c42f, 2015-12-13:
Seems like this is not entirely deterministic - selecting something other than the first in the list sometimes also causes a crash, and the procedure above doesn't always reproduce the bug.

landersson, 2016-02-23:
Hi Chris, 

I also just ran into exactly this problem while modifying displaz to add a new remote command for removing datasets. I'm running Ubuntu 14.04 with NVIDIA driver 352.79 using a GTX 670. Let me know if you figure out what's going on. My knowledge of modern OpenGL is probably not up to scratch for debugging this further.

Anyway, thanks for a very useful program and nicely hackable code... I've found displaz invaluable for visualizing and interacting with large point cloud data sets in combination with .ply vector data.

c42f, 2016-02-23:
I first noticed this right after the Qt5 port which makes me think it's an issue with datasets being accessed after deletion by the UI list code, or some such thing (potentially unrelated to OpenGL, but segfaulting there)...  Indeed it's a nasty and tricky bug to track down, and trying to reproduce it on my work computer just now failed :-(  Will try again later at home.

I'm glad you're finding displaz useful!  I plan to continue adding remote commands to make the UI as scriptable as possible, so removing data sets remotely is definitely within scope.  My current plan is first to make it unnecessary in many cases, by making the `-dataname` option replace an existing data set with the same name by default.  We'd also want a way to refer to datasets by a unique ID to allow for more complex manipulation (this may be the name, or it may just default to the name, I'm not sure.  See the dataset-id-interface branch where I started down the path of having a uuid per dataset.  This turned out to be a bit messy and rather too chatty over the socket, so I don't plan to merge it in the current state, but it's a potentially interesting point in the design space.)

c42f, 2016-02-23:
By the way, if you're planning to add new commands and you'd like to contribute them, please open an issue and we can discuss the best design.  In the future I'd like to officially document the socket protocol, but only after it's made into something which sucks a lot less ;-)  Something like a json or msgpack based command language would probably make more sense than the mess I currently have :)

landersson, 2016-02-23:
Not thoroughly investigated, but commenting out "destroyBuffers()" on line 54 in geometry.cpp seems to stop my test case (removing file 2 in a list of 3 files (.las, .ply, .ply)) from crashing.

Running displaz-gui under valgrind does not report any illegal accesses by the UI list code... all I'm seeing is a null pointer access in the nvidia driver triggering a segfault.

Happy to contribute any new commands that I'll be adding. So far I added a "-viewposition %F %F %F" command to move the viewpoint to a new absolute position, and was attempting to add a "-remove %S" to remove a dataset by name when running into the segfault issue. I'll fork displaz and send you a link once I've had time to clean it up a bit.

c42f, 2016-02-23:
Excellent, thanks.

Regarding this bug, turning on OpenGL error checking (see `GL_CHECK` in glutil.h) reveals several problems which I'll need to dig through.  They may not be the cause of the problems here, but should be fixed.  (Eh, sorry for the breakage, ideally I would have tested the Qt5 port more thoroughly before merging it a few months ago.)

Via valgrind I also noticed that the usage of `glDeleteFramebuffers()` is totally bogus at the moment, I'll need to fix that, but time for bed now...

commit ed06bb4b (2016-07-01): Ensure vertex array is unbound after draw

Vertex arrays must be unbound after drawing.  If not, OpenGL's evil
global state can leak into later functions, in some cases causing a
segfault in the driver when the bound CPU buffers have been deallocated.

Fixes #86
closed 2016-07-01 by c42f