c42f/displaz #87: Build failure on ubuntu-15.10

c42f, 2016-01-07:
The build fails on ubuntu 15.10, with qt headers complaining about `-fPIE`, which overrides `-fPIC` in the compile flags.  It's not very clear where the `-fPIE` is coming from (possibly cmake?)

juliohm, 2016-01-30:
@c42f I have the exact same issue on Arch Linux. Is there any workaround?

This actually may be an issue with the Qt 5 CMake modules, I also couldn't figure out where the -fPIE is coming from.

juliohm, 2016-01-30:
Google search returned a few interesting links:

* `Qt 5.4.2` important behavior changes: http://code.qt.io/cgit/qt/qtbase.git/plain/dist/changes-5.4.2
* Another attempt to fix it: https://github.com/Cockatrice/Cockatrice/commit/9cfb09cd5b25a01fb142d425ba70b77b89f65a09

I tried both, but they didn't work. I am using `GCC 5.3.0` and `CMake 3.4.2`.

c42f, 2016-01-30:
There's a workaround, but it's pretty disgusting ;-)

Build displaz according to the instructions, but replace the make step with 

```sh
make VERBOSE=1
```

This will spit out full information about each build step.  When you get to the failing step (IIRC it's a linker step), just copy the compile command which fails, remove the offending -fPIE and run it by hand.  The rest of the build should then go through, but if not, rinse and repeat...

Come to think of it, you could probably munge the makefiles using grep and sed which would make things slightly less manual.  How about trying something like the following from inside the build directory:

```sh
sed -i -e 's/\<fPIE\>/fPIC/g' $(grep -l -R -e '-fPIE')
```

then run make as usual.  Be aware that every time cmake is run you'll need to repeat this step before running make.

juliohm, 2016-01-30:
That is indeed a hack :blush:. For me the error happens when trying to compile `typespec.cpp`, it includes the header `glutil.h` which apparently include Qt headers.

After I remove `-fPIE` for this file, the build is fine. I opened `displaz-gui` and got this log:

[image]

Is everything ready for use or I need to install other missing dependencies?

juliohm, 2016-01-30:
It is fine, I had to go all the way through and install `make install` to have the `share` files in place. The screenshot above only shows the error because I tried to run from the build directory.

c42f, 2016-01-31:
Correct, it doesn't run properly from the build directory.  I used to build the shaders into the executable as resources and this did work.  However, having them in their own files in the install directory is a lot more transparent and hackable for users, so now displaz does things that way.  (I encourage people to fiddle with the shaders themselves to achieve custom visualizations.)

c42f, 2016-01-31:
By the way, do you have the list of package dependencies required for building displaz on Arch?  I'd be happy to add them to the readme instructions if you do (or you could submit a PR for it ;-) )

juliohm, 2016-01-31:
Except for that issue with `-fPIC`, the build was smooth, very well written `CMakeLists.txt` by the way :+1:. I didn't install anything else to get it working.

c42f, 2016-01-31:
Thanks.  Yeah I suspect the `fPIE` problem is actually a cmake bug or obscure misconfiguration issue.  I haven't had time to track it down yet.

c42f, 2016-01-31:
The underlying issue looks to be related to [CMP0018](https://cmake.org/cmake/help/v3.0/policy/CMP0018.html) and a bad interaction with the way that Qt wants compiler flags to be imported.  Looks like it's a bit of a mess if you're trying to support various cmake and qt versions.

nigels-com, 2016-05-07:
Same problem on Ubuntu 16.04 with gcc 5.3.1 and cmake 3.5.1.

nigels-com, 2018-12-28:
Ubuntu 15.10 is past end of life.  I can confirm that both 14.04 and 16.04 are fine.  I think this issue can be closed now.

https://www.ubuntu.com/about/release-cycle