c42f/displaz #90: Tabulated LiDAR format

juliohm, 2016-01-30:
I never played with LiDAR before, but I got this data split into 4 files:

`R-channel.dat`
>x1 y1 R1
>x2 y2 R2
>...

`G-channel.dat`
>x1 y1 G1
>x2 y2 G2
>...

`B-channel.dat`
>x1 y1 B1
>x2 y2 B2
>...

`z-coordinate.dat`
>x1 y1 z1
>x2 y2 z2
>...

At least that is my interpretation of the numbers and file names. What is the best way to visualize this data in Displaz?

c42f, 2016-01-31:
Hmm, this seems like a strange data layout, I've seen datasets with channels in separate files before and it always surprises me why anyone would want to do this :-)  I assume the `x1 y1` in the first file is the same `x1 y1` in the other files?

At the moment, the displaz native format is ply, so the easiest way to see the data is to turn your files into the version of ply that displaz likes to read.  There's various ways to do this, but from your profile I see you're likely using the julia bindings in Displaz.jl so that would probably be easiest for you.  Off the top of my head, the appropriate script would look something like

```julia
using Displaz

# Assumes x y coordinates in all files are shared

position = readdlm("z-channel.dat")' # all of x,y,z

# Read color, discarding x,y
R = readdlm("R-channel.dat")[:,3]
G = readdlm("G-channel.dat")[:,3]
B = readdlm("B-channel.dat")[:,3]

color = [R G B]'

plot3d(position, color=color)
```

juliohm, 2016-01-31:
Yes, this is a weird format they gave me. I also don't understand the rationale in separating the channels. I merged the files and converted with `txt2las -parse xyzRGB -i foo.dat` from the LAStools package. Apparently it is not working as expected because the visualization in displaz doesn't show any color. I will try the ply format as you suggested, thanks for sharing the Julia script :blush:, I will definitively check `Displaz.jl`.

c42f, 2016-01-31:
The reason for the lack of color is that displaz doesn't have a flexible text file parser: it assumes the first three colums are xyz and ignores the rest of the columns.  I'd like to make it better, but I'd need some systematic way for the user to inject metadata about the meaning of the columns, or a way to guess.

Delimited text is a bit of an infuriating format: on the one hand, it's so simple that it makes you want to use it; on the other hand there's not really the possibility of standard metadata.  I could suggest that people add column names on the first line, and try to guess the meaning of the columns from those.  It would certainly work for a set of the most common things (xyzrgb)

juliohm, 2016-01-31:
I think the issue is actually the `txt2las` converter. I didn't load text files directly into Displaz, I loaded the generated `*.las` file and wasn't able to see the colors.

juliohm, 2016-01-31:
The Displaz script you posted solved the issue, thank you!

closed 2016-01-31 by juliohm

c42f, 2016-01-31:
Cool.  Of course the nice thing about going via julia is that the color is entirely user defined so you can use it for many other things as you work with the data.

`txt2las` probably does do what you want - I believe lastools has a rather large install base and people would quickly notice if it was broken.  What's more likely is that your RGB channels are scaled from 0 to 255, rather than 0 to `2^16-1`, so they appear black.  You can fix that by just increasing the `exposure` shader control using the UI.

c42f, 2016-01-31:
(The `2^16-1` scaling is what's specified in the las standard, but I could probably insert a hack to rescale the color in the case that people have non-conforming las files.  I'm unsure whether that would be too magical though... setting the exposure automatically would probably make more sense.)

juliohm, 2016-01-31:
In this case the channels have values in the interval you mentioned `[0, 2^16-1]`, I checked the *.las generated with `txt2las` by applying the inverse `las2txt`. It seems that the decimals are being truncated in the process for some reason. Anyways, this is not very relevant now that you suggested this more convenient way with Displaz.jl.

c42f, 2016-01-31:
ok... mysterious, but no worries.  Definitely works for some colored las I've got sitting around here!