c42f/displaz #91: Screenshot with undesired menu box

juliohm, 2016-01-31:
When we take a screenshot, this is how it is saved (PNG):

[image]

c42f, 2016-01-31:
Yes, this is definitely a bug, but fixing it properly wasn't easy last time I looked.

As a very easy workaround, you can press F9 directly without using the menu.  You may also turn off any unwanted decorations using the View menu.

nigels-com, 2016-01-31:
Yeah, it's better to render offscreen for nice results with images.

c42f, 2016-01-31:
@nigels-com - there's even supposedly a function in QGLWidget to do exactly that, but it forces a different QGLContext, which led to strange crashing behaviour last time I tried and I wasn't sufficiently motiviated to fix it.  May be better/easier in Qt5, I haven't looked.

c42f, 2016-01-31:
Actually this is probably quite a lot easier to fix than last time I thought about it, with the points already rendered to an offscreen buffer internally.

Side note - it should also be possible to optionally render screenshots with antialiasing, which would make them a lot nicer in many cases such as this one.

nigels-com, 2016-01-31:
FBOs might be nice for this.  For a stationary camera you could skip all the heavyweight rendering and do annotations, labels and so as a compositing-style step.

c42f, 2016-01-31:
I guess that's kind of how it works already with the bounding boxes, cursor and axes drawn onto the backbuffer after blitting in the points which are rendered offscreen in multiple passes.

juliohm, 2017-01-17:
What is the status of this issue?

c42f, 2017-01-21:
It's still open, and I'm not sure when I'll get to it.  Alas I don't have a lot of free time right now.

Luckily, as stated above there's a very simple workaround: just use the keyboard shortcut.