c42f/displaz #92: Hookable events for scripting interfaces

c42f, 2016-02-04:
For automating displaz from julia/python/etc it'd be really useful to be able to register keyboard/mouse event hooks from the scripting language and have data returned on each event.  This would make it possible to round trip from scripting language to form simple 3D UIs while prototyping algorithms.

Usage sketch:

1. Scripting language sends a hook command down the socket via the displaz CLI, eg `displaz -hookevent key:E cursorposition`, and sets up a task/thread blocking on reading the displaz process stdout.
2. displaz process waits on the socket
3. When displaz-gui reads the E key, it responds with the cursor position, which is read by displaz and sent to stdout
4. Script reads a line from stdout with the desired message, and responds by sending extra datasets back to displaz-gui

The hook in displaz-gui should disappear when the displaz CLI process exits.

The model above is pretty simple, but may be a bit command line centric and doesn't seem extensible to complex messages.  Possibly might want to consider a proper serialization format for socket messages rather than the ad hoc line-based message format.

c42f, 2016-02-16:
Additional useful hook event:  When user clicks on a point, return (dataset name, vertex index, vertex attributes) - more or less what currently goes in the log

landersson, 2016-03-17:
I nearly started writing something along these lines a little while ago when I wanted displaz-gui events to trigger actions in my python script. The way I planned to do it was to somehow give displaz-gui an ip+port to log events to, and have my python script open up a simple tcp server to record those. Your approach might be more straightforward. I ended up doing something less user friendly to get the task at hand done, but being able to pass displaz events back to a script is something I'd find quite useful.

c42f, 2016-03-17:
Internally displaz does something very much like that for its IPC, but with a local unix domain socket rather than a network socket.  For security reasons, I don't particularly want to harden displaz against external programs attacking it via the network.

Scripting language bindings can talk directly down the IPC socket in principle, but then each socket interface has to handle a few annoying things:

* If an existing displaz instance isn't found (nothing answers on the socket), either fail, or figure out how to open one.
* Deal with async socket IO; this involves varying levels of badness depending on the scripting language.
* The socket API isn't currently public.  We'd need a less hacky serialization format before making it so (json? msgpack?)

This is not to say I wouldn't consider a direct socket connection in the future, but it does come with some caveats which are currently handled in one place (ie, the displaz command line interface C++ code).

janzill, 2016-06-02:
I thought the pull request message would show up here but it doesn't, so here's a copy to discuss, sorry for the duplicate: 
 I had a crack at issue #92 and implemented a simple CLI protocol along the lines of your outline, Chris. There are a couple of things I'm not entirely sure about:
 - Is the CLI supposed to wait for stdin after a single message is sent from the GUI, or would it be better to keep the hook active while no input is incoming (i.e. getting multiple events back)?
 - What kind of information is the script supposed to communicate back to the GUI, and what does the GUI do with it?
 - Do you want to be able to specify a key and event message to be returned to the CLI?

 Regarding the first point, the CLI is currently sending a single message to stdout and then waits for input from the script. If displaz would create another thread for the input and wait until it receives input, it would be possible to get as many events as desired with a single hook (the output would have to be blocked as soon as input starts).
 Regarding the second point, if there are complex messages to be sent back and forth a serialization of the socket communication messages would make the implementation easier as you mentioned.
 Regarding the third point, an event handler in a separate class might make sense here (this would also make View3D a bit smaller).

For now I implemented a single option (-hook) without arguments, and three events with associated behaviour (for reference: key:E returns coordinates of cursor position; right click+Control (command on Mac) returns position coordinates of closest point to click; right click+Shift returns more information on closest point to click ).  A combination of event handler and socket communication would make this easily extendable.

c42f, 2016-06-03:
Ah yes, github doesn't integrate issues and pull requests very closely, you just get a reference to the other generated each time you mention it (eg #92).

As a first implementation I'd be happy to continue avoiding direct socket communications, in my experience so far it's somewhat painful to get it working consistently across platforms.

c42f, 2016-06-14:
Done in #108

closed 2016-06-14 by c42f