c42f/displaz #97: Screenshot Command

landersson, 2016-03-16:
I'm aiming to use displaz to render a simple point cloud fly through sequence and would like to add a "screenshot %s" remote command to save the current view to an image file. One complication I see is regarding the incremental drawing. Any input on the proper way to temporarily disable incremental drawing and make sure all points have been rendered before saving the frame would be appreciated.

c42f, 2016-03-16:
This makes sense, I'd like to have a `-screenshot` remote command.

The guts of this are in `src/render/View3D.cpp` in `View3D::paintGL()` - I'd suggest refactoring this a bit to get what you want.  All points are currently drawn to an offscreen framebuffer `m_incrementalFramebuffer`; lines, meshes and bounding boxes are also drawn here at the moment, but only on the first incremental frame, since I don't have a way to render them incrementally.  The current incremental buffer is then blitted into the backbuffer each frame, followed by rendering the overlay stuff like the cursor (arguably this per-frame rendering should also include bounding boxes and other cheap annotations).

I think we should refactor `paintGL()` so that the important parts are pulled out and can be called independently of the current GUI state - that is, into static or at most const helper functions.  This would also allow the screenshot rendering to be performed to a separate offscreen framebuffer (rather than `m_incrementalFramebuffer`, you could even turn on antialiasing to make things prettier.  When doing a screenshot, you'd want to pass `quality = DBL_MAX` to the helper function, along with (possibly?) a custom camera position, etc.

landersson, 2016-03-21:
Just sent #98.