c42f/displaz #99: Windows build fails

JoshChristie, 2016-04-01:
After installing the continuous windows build (https://displaz-data.s3.amazonaws.com/builds/ci/latest/displaz-win64.exe), I get the following error when I try to run the executable. 
[image]

nigels-com, 2016-04-01:
Yes, confirming the same problem here on Windows 7, 64-bit.  The workaround that seems to work for me is to use some Qt dlls from a previous install of displaz.

c42f, 2016-04-04:
Looks like this is due to Qt making more things into plugins so they can't be found automatically via `fixup_package()`.  Relevant links

http://stackoverflow.com/questions/31920698/how-to-deploy-qwindowsintegrationplugin-with-cmake

Looks like there's some relevant cmake song and dance in GCompris-qt 
https://github.com/bdoin/GCompris-qt/blob/e88ea239056c54410451cc5da660ad264dd7ae37/src/core/CMakeLists.txt#L96
https://github.com/bdoin/GCompris-qt/blob/a33ff0300ce352e7b727e0c8e39b6cb5ebd16947/cmake/qt_helper.cmake#L48

Ick :-(

c42f, 2016-06-10:
Another useful link

https://gitlab.kitware.com/kotu/MoleQueue-Qt5.0.2/blob/a1dcf93ed46a7c7a101ae448f912a3351679d3cf/cmake/DeployQt5.cmake

c42f, 2016-06-14:
A junky workaround for this looks something like

```
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index 80378f5..06340d8 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -231,6 +231,9 @@ if (DISPLAZ_BUILD_DVOX)
     install(TARGETS dvox DESTINATION "${DISPLAZ_BIN_DIR}")
 endif()
 
+get_property(loc TARGET Qt5::QWindowsIntegrationPlugin PROPERTY LOCATION_RELEASE)
+install(FILES ${loc} DESTINATION bin/plugins)
+
 if(WIN32)
     # Install required qt libraries
     if (DISPLAZ_USE_QT4)
```

c42f, 2017-05-12:
Fixed, thanks to @evetion in #168

closed 2017-05-12 by c42f