c42f/displaz #111: Dynamic in-opengl-window UI elements.

roames, 2016-06-15:
To move away from the Qt-style UI that mostly occupies precious viewport space (but is good for displaying logs), add a dynamic (ideally scriptable) UI that renders as part of the OpenGL window. It could be used to develop small script plugins (#92) for Displaz that would come with their own UI components.

Measurement tools (#53, #9) could be included this way, so could configurable colourmaps (#44).

Possible UI lib examples (in order of personal preferences):
 1. [NanoUI](https://github.com/wjakob/nanogui) - mainly for it's support of awesome widgets (for image,function,colour,etc display)
 2. [AntTweakBar](http://anttweakbar.sourceforge.net/doc/) - much older and more stable

c42f, 2016-06-16:
As I said the other day, this sounds interesting.  I'd be particularly interested if it would eventually allow the qt UI to go away entirely, in which case we'd need the "awesome widgets" version :-)

I'd be interested to see a small prototype replacing a portion of the current UI before anyone embarks on a full scale rewrite of things.

nigels-com, 2016-06-16:
NanoUI needs some work to decouple of GLFW.  Doesn't seem like a major effort, just an abstract interface with optional GLFW back-end ought to work.  However, NanoUI doesn't seem to have anything for multi-line text editing, which seemed like a key UI element to me.

r-chris, 2016-06-16:
We would have to fork from the main NanoUI repo, because it doesn't seem like substantial changes related to GLFW will be accepted upstream by the author.

@nigels-com you think we absolutely need text editing (presumably for shaders) or maybe we should go down the path of using an external editor and just reload the files internally. That proposal appears to have been considered already in issue #13 and would also make sense for #43

nigels-com, 2016-06-16:
Has the author specifically said they wouldn't accept an alternative back-end to GLFW?  I'd certainly prefer to have a console and/or shader editor, but I'm not sure there is an off-the-shelf solution for that.

r-chris, 2016-06-16:
Building our own shader editor (ideally with syntax highlighting, etc) seems counterproductive in my opinion. There are so many great text editors out there that we use everyday - I would prefer to allow anyone to use the editor of their preference over some featureless ugly text box. It doesn't strike me as something that really should be part of Displaz.

r-chris, 2016-06-16:
In relation to getting rid of GLFW and other external dependencies this is the authors common response:

> I'd like to keep the dependencies there to make building the project absolutely straightforward.
> [...]
>  I do, however, accept patches if they contribute useful functionality or architectural improvements.

However, when I submitted a PR that was even hinting at getting rid of GLFW he said this:

> Sorry, I can't merge the PR in this form. The GLFW used by nanogui is based on a repository with build system modifications, so it's not straightforward to just use a different GLFW library.

nigels-com, 2016-06-17:
You think @miyax is unlikely to accept a patch to opt-out of GLFW and opt-in to Qt?  In that case we ought to look elsewhere.

r-chris, 2016-06-17:
That appears to be the author of an abandoned library called [nanoui](https://github.com/miyax/nanoui) which is also based on NanoVG, but not the NanoGUI we are looking at right now (I misspelled it in the original post - sorry about that).

The author would be @wjakob, but do we really want to keep using Qt here?

c42f, 2016-06-17:
I'm personally not married to qt for displaz, but the source code is!  It'd be quite a bit of work to port it in one big lump and we'd need to depend on some extra libraries to replace QtCore functionality.

Personally I think iterative approach to improving the UI seems more practical here rather than changing everything at once.

And before we do anything, I'd like to see a prototype of the kinds of things you'd like to achieve, just to convince me it's worth it ;-)

nigels-com, 2016-06-17:
Indeed, I don't think Qt will be disappearing from displaz anytime soon, even if we decide on a really good solution for OpenGL pointy-clicky.

r-chris, 2016-06-17:
Johann Wolfgang von Goethe:
> Der Worte sind genug gewechselt, Laßt mich auch endlich Taten sehn!

loosely translates to

> Enough words have been exchanged, let me finally see the deeds!

nigels-com, 2016-06-17:
All the risk and all the glory, huh?

bhack, 2017-06-06:
Have you evaluated qt3d in 5.9?

c42f, 2017-06-07:
Not yet.  I assume you refer to https://blog.qt.io/blog/2017/05/24/qt3d/ ?  To quote... "embedding Qt Quick into a Qt 3D scene is now easily possible".  This is a huge step forward.

For background, displaz started when qt-5.0 was still not released, and even in early qt-5 versions, the support for OpenGL features we needed was still lacking so there's a bunch of home grown tools in the source. Qt OpenGL tools have come a very long way since then.

bhack, 2017-06-07:
Yes I'm referring to that. Probably you will got also Vulkan for free soon or later https://blog.qt.io/blog/2017/06/06/vulkan-support-qt-5-10-part-1/

closed 2018-02-28 by r-chris