c42f/displaz #133: Modify point attributes in place using index lists

c42f, 2016-09-12:
For interactive scripts which use displaz as a 3D window, it would be very useful to be able to modify the properties of a subset of vertices in an existing dataset, especially if that dataset is quite large.

An implementation idea is to pass a dataset label, a set of indices and set of attributes to modify, with displaz then making the modifications in place.  The indices should be the indices of vertices loaded from the initial point cloud file (as opposed to the rearranged ordering as stored in memory).

Versioning on the displaz side could be supported if desired, by storing the previous attribute values - this would make implementing undo systems in scripts far easier.

@andyferris - here's an issue to allow modifying point clouds in place.

c42f, 2016-09-12:
Implementation notes:

Potential command line usage:

```
displaz -modify <my_geom_label> <index_and_attributes>.ply
```

* The "displaz ply" format as it exists is a viable input format for the index list and attributes.
* The FileLoader thread currently only loads instances of `Geometry`, but for modifying an existing Geometry we should produce a `GeometryMutator` or some such thing, which gets applied to an existing geometry on the main thread.
* `-modify` could be `-mutate`, `-update`, ... ?

andyferris, 2016-09-12:
Seems like a really cool idea Chris; check out a small start at #134.

c42f, 2016-09-25:
Done in #134

closed 2016-09-25 by c42f