c42f/displaz #136: Idea: Scripting access to logger

andyferris, 2016-09-16:
At the moment there are few options that a scripting/hooks application can interact with displaz's user. One simple way would be to allow a scripting command for adding a log entry, like:

```
displaz -loginfo "Welcome to my app"
displaz -logwarning "You did the wrong thing"
displaz -logerror "You REALLY did the wrong thing"
```

These commands would be able to be used with or without a command to load a file or mutate some data. For completeness, to match the Qt logger, we could also implement `-logdebug`, `-logcritical` and `-`logfatal` (though I'm not sure whether or not the last should apply if its just the scripting app that dies).

c42f, 2016-09-16:
Yes, this would be good.  I'd probably go for a single command line option, perhaps

```
displaz -notify <level> <message>
```

would be clearer naming?   `displaz -notify error "You are a bad, bad user."`

andyferris, 2016-09-16:
+1

Perhaps in the future there may be more ways to interact with the user, e.g. through the log or through a pop-up dialogue or a widget in the OpenGL window or whatever. I'm wondering if "notify" the correct word to indicate specifically the log?

andyferris, 2016-09-16:
I guess these could be 

```
displaz -notify popup "You made me angry, and now I will make you angry with this pop-up notification"
```

or whatever

c42f, 2016-09-16:
lol.

Yes, having a string as the notification type is pretty extensible (could have a simple format `$style:$severity`, for example: `popup:error` `log:error` `status:info`).  Though a popup dialog style is unlikely to ever be merged :-P

c42f, 2017-02-24:
Closed by #148

closed 2017-02-24 by c42f