c42f/displaz #139: Semi-transparent polygons

c42f, 2016-09-29:
Crude support for semi-transparent polygons would be quite useful  CC @Ross-Batten .

In general, there's no nice way to do this in standard OpenGL, since OpenGL doesn't support order independent transparency; without special CPU-side handling, rendering artifacts are guaranteed.  As far as I'm aware, the usual solution to this is:

1. Render all opaque geometry as usual
2. Ensure transparent primitives cannot intersect using high level logic ("game logic")
3. Sort transparent primitives by depth
4. Render primitives in the sorted order, back to front, with alpha compositing enabled.

We could potentially implement the above in a crude way, by just ignoring step 2 which can't be fixed without heroic efforts.  The idea would be:

* Add transparency flag to `Geometry` instances.
* Separate transparent `Geometry` instances into a separate list, sort, and render front to back.

nigels-com, 2016-09-29:
_Depth Peeling_ is a potential approach, depending on the number of layers.
Fairly GPU intensive.  https://en.wikipedia.org/wiki/Depth_peeling
http://doc.jzy3d.org/depthpeeling/DualDepthPeeling.pdf

c42f, 2016-09-30:
Thanks, this is interesting - particularly the single pass approximation - displaz is not aiming for realism but utility, so using an approximation would be fine as long as it's not distracting.

bhack, 2017-06-06:
Probably something could be done in the [QT3D framegraph](https://indico.cern.ch/event/617054/contributions/2519723/attachments/1435357/2206891/cernhfs-qt3d-basics.pdf).