c42f/displaz #140: Compile error

svHatch, 2016-09-30:
Error when I got to the internal build make statement:

[  1%] Scanning dependencies of target embeddedGLEW
Scanning dependencies of target doc
Scanning dependencies of target InterProcessLock_test
Generating moc_IpcChannel.cxx
[  2%] Generating README.html
[  3%] Building C object CMakeFiles/embeddedGLEW.dir/thirdparty/glew/glew.c.o
Scanning dependencies of target displaz
[  4%] Building CXX object src/CMakeFiles/InterProcessLock_test.dir/InterProcessLock_test.cpp.o
[  5%] Generating userguide.html
[  6%] Building CXX object src/CMakeFiles/displaz.dir/logger.cpp.o
[  6%] Built target doc
[  7%] Building CXX object src/CMakeFiles/displaz.dir/geomfield.cpp.o
[  8%] Building CXX object src/CMakeFiles/displaz.dir/typespec.cpp.o
[  9%] Building CXX object src/CMakeFiles/InterProcessLock_test.dir/util.cpp.o
[ 10%] Building CXX object src/CMakeFiles/displaz.dir/hcloud.cpp.o
[ 11%] Building CXX object src/CMakeFiles/InterProcessLock_test.dir/InterProcessLock.cpp.o
[ 12%] Building CXX object src/CMakeFiles/displaz.dir/util.cpp.o
[ 13%] Building CXX object src/CMakeFiles/displaz.dir/__/thirdparty/argparse.cpp.o
Linking CXX executable ../bin/InterProcessLock_test
[ 13%] Built target InterProcessLock_test
Scanning dependencies of target unit_tests
[ 14%] Building CXX object src/CMakeFiles/displaz.dir/moc_IpcChannel.cxx.o
[ 15%] Building CXX object src/CMakeFiles/unit_tests.dir/logger.cpp.o
[ 17%] Building CXX object src/CMakeFiles/displaz.dir/InterProcessLock.cpp.o
[ 18%] Building CXX object src/CMakeFiles/displaz.dir/IpcChannel.cpp.o
[ 19%] Building CXX object src/CMakeFiles/unit_tests.dir/geomfield.cpp.o
Linking C static library lib/libembeddedGLEW.a
[ 19%] Built target embeddedGLEW
[ 20%] Generating qrc_resource.cxx
[ 21%] Generating moc_fileloader.cxx
[ 22%] Building CXX object src/CMakeFiles/displaz.dir/main.cpp.o
[ 23%] Generating moc_geometrycollection.cxx
[ 24%] Generating moc_HookFormatter.cxx
[ 25%] Generating moc_HookManager.cxx
[ 26%] Generating gui/moc_DataSetUI.cxx
[ 27%] Generating gui/moc_DragSpinBox.cxx
[ 28%] Generating gui/moc_HelpDialog.cxx
[ 29%] Generating gui/moc_InteractiveCamera.cxx
[ 30%] Generating gui/moc_PointViewerMainWindow.cxx
[ 31%] Generating gui/moc_QtLogger.cxx
[ 32%] Generating gui/moc_ShaderEditor.cxx
[ 34%] Generating render/moc_Geometry.cxx
[ 35%] Generating render/moc_HCloudView.cxx
[ 36%] Generating render/moc_View3D.cxx
[ 37%] Generating render/moc_PointArray.cxx
[ 38%] Generating render/moc_Shader.cxx
[ 39%] Building CXX object src/CMakeFiles/unit_tests.dir/typespec.cpp.o
[ 40%] Generating render/moc_GeometryMutator.cxx
[ 41%] Building CXX object src/CMakeFiles/unit_tests.dir/hcloud.cpp.o
Scanning dependencies of target displaz-gui
[ 42%] Building CXX object src/CMakeFiles/displaz-gui.dir/logger.cpp.o
Linking CXX executable ../bin/displaz
[ 42%] Built target displaz
[ 43%] Building CXX object src/CMakeFiles/displaz-gui.dir/geomfield.cpp.o
[ 44%] Building CXX object src/CMakeFiles/unit_tests.dir/util.cpp.o
[ 45%] Building CXX object src/CMakeFiles/unit_tests.dir/__/thirdparty/argparse.cpp.o
[ 46%] Building CXX object src/CMakeFiles/displaz-gui.dir/typespec.cpp.o
[ 47%] Building CXX object src/CMakeFiles/displaz-gui.dir/hcloud.cpp.o
[ 48%] Building CXX object src/CMakeFiles/unit_tests.dir/streampagecache_test.cpp.o
[ 50%] Building CXX object src/CMakeFiles/unit_tests.dir/util_test.cpp.o
[ 51%] Building CXX object src/CMakeFiles/displaz-gui.dir/util.cpp.o
[ 52%] Building CXX object src/CMakeFiles/displaz-gui.dir/__/thirdparty/argparse.cpp.o
[ 53%] Building CXX object src/CMakeFiles/displaz-gui.dir/moc_IpcChannel.cxx.o
[ 54%] Building CXX object src/CMakeFiles/unit_tests.dir/test_main.cpp.o
[ 55%] Building CXX object src/CMakeFiles/displaz-gui.dir/InterProcessLock.cpp.o
[ 56%] Building CXX object src/CMakeFiles/displaz-gui.dir/IpcChannel.cpp.o
[ 57%] Building CXX object src/CMakeFiles/displaz-gui.dir/moc_fileloader.cxx.o
[ 58%] Building CXX object src/CMakeFiles/displaz-gui.dir/moc_geometrycollection.cxx.o
[... 62 lines cut ...]
Building CXX object src/CMakeFiles/displaz-gui.dir/gui/PointViewerMainWindow.cpp.o
Building CXX object src/CMakeFiles/displaz-gui.dir/render/PointArray.cpp.o
/home/scott/bin/displaz/src/gui/PointViewerMainWindow.cpp: In member function ‘void PointViewerMainWindow::handleMessage(QByteArray)’:
/home/scott/bin/displaz/src/gui/PointViewerMainWindow.cpp:479:57: error: invalid conversion from ‘float*’ to ‘int’ [-fpermissive]
         m_pointView->camera().setRotation(QMatrix3x3(rot));
                                                         ^
In file included from /usr/include/qt4/QtGui/qmatrix4x4.h:48:0,
                 from /usr/include/qt4/QtOpenGL/qglshaderprogram.h:49,
                 from /usr/include/qt4/QtOpenGL/QGLShader:1,
                 from /home/scott/bin/displaz/src/render/Shader.h:10,
                 from /home/scott/bin/displaz/src/gui/PointViewerMainWindow.cpp:15:
/usr/include/qt4/QtGui/qgenericmatrix.h:106:5: note: initializing argument 1 of ‘QGenericMatrix<N, M, T>::QGenericMatrix(int) [with int N = 3; int M = 3; T = double]’
     QGenericMatrix(int) {}       // Construct without initializing identity matrix.
     ^
/usr/include/qt4/QtGui/qgenericmatrix.h:106:5: error: ‘QGenericMatrix<N, M, T>::QGenericMatrix(int) [with int N = 3; int M = 3; T = double]’ is private
/home/scott/bin/displaz/src/gui/PointViewerMainWindow.cpp:479:57: error: within this context
         m_pointView->camera().setRotation(QMatrix3x3(rot));
                                                         ^
src/CMakeFiles/displaz-gui.dir/build.make:959: recipe for target 'src/CMakeFiles/displaz-gui.dir/gui/PointViewerMainWindow.cpp.o' failed
make[2]: *** [src/CMakeFiles/displaz-gui.dir/gui/PointViewerMainWindow.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:221: recipe for target 'src/CMakeFiles/displaz-gui.dir/all' failed
make[1]: *** [src/CMakeFiles/displaz-gui.dir/all] Error 2
Makefile:147: recipe for target 'all' failed
make: *** [all] Error 2

nigels-com, 2016-09-30:
Yes, confirming this on Ubuntu 16.04 with `-DDISPLAZ_USE_QT4=TRUE`

```
/home/nigels/dev/displaz/src/gui/PointViewerMainWindow.cpp: In member function ‘void PointViewerMainWindow::handleMessage(QByteArray)’:
/home/nigels/dev/displaz/src/gui/PointViewerMainWindow.cpp:479:57: error: invalid conversion from ‘float*’ to ‘int’ [-fpermissive]
         m_pointView->camera().setRotation(QMatrix3x3(rot));
                                                         ^
In file included from /usr/include/qt4/QtGui/qmatrix4x4.h:48:0,
                 from /usr/include/qt4/QtOpenGL/qglshaderprogram.h:49,
                 from /usr/include/qt4/QtOpenGL/QGLShader:1,
                 from /home/nigels/dev/displaz/src/render/Shader.h:10,
                 from /home/nigels/dev/displaz/src/gui/PointViewerMainWindow.cpp:15:
/usr/include/qt4/QtGui/qgenericmatrix.h:106:5: note:   initializing argument 1 of ‘QGenericMatrix<N, M, T>::QGenericMatrix(int) [with int N = 3; int M = 3; T = double]’
     QGenericMatrix(int) {}       // Construct without initializing identity matrix.
     ^
/usr/include/qt4/QtGui/qgenericmatrix.h:106:5: error: ‘QGenericMatrix<N, M, T>::QGenericMatrix(int) [with int N = 3; int M = 3; T = double]’ is private
/home/nigels/dev/displaz/src/gui/PointViewerMainWindow.cpp:479:57: error: within this context
         m_pointView->camera().setRotation(QMatrix3x3(rot));
                                                         ^
src/CMakeFiles/displaz-gui.dir/build.make:1001: recipe for target 'src/CMakeFiles/displaz-gui.dir/gui/PointViewerMainWindow.cpp.o' failed
make[2]: *** [src/CMakeFiles/displaz-gui.dir/gui/PointViewerMainWindow.cpp.o] Error 1
```

nigels-com, 2016-09-30:
I'd suggest trying Qt5 with  `cmake .. -DDISPLAZ_USE_QT4=FALSE`

c42f, 2016-09-30:
Yes, this the first sign of the qt4 support bitrotting as I don't build with it any more.  (Looks like I've accidentally used a constructor for `QMatrix4x4` which doesn't exist in qt4.)

As Nigel says, if you have the option of building against qt5 this should fix the problem (do you have a system which only has qt4 packages?)

As a side note, I think the next release (quite soon I hope) will be the last one to support qt4, as it's pretty much abandoned by the qt developers these days.

c42f, 2016-10-01:
Should be fixed in de7803c

closed 2016-10-01 by c42f