c42f/displaz #147: Text annotations in 3D window

c42f, 2016-10-12:
To annotate objects via the scripting bindings, it would be really useful to have arbitrary billboarded text in the 3D window.

The simplest way to communicate this from the scripting bindings would probably be to have text passed directly from the command line, eg, `displaz -text $x $y $z "Some text"`.

CC @Ross-Batten

c42f, 2016-10-14:
Valve's signed distance field technique for font rendering seem to be the current sweet spot of good looking and practical for text rendering. It would be a lot of work, but luckily the qt libraries already have some of it done.  Unfortunately it's part of QtQuick 2, so it might be hard to integrate
https://www.kdab.com/integrating-opengl-with-qt-quick-2-applications-part-1/

nigels-com, 2016-10-15:
If we can live without zooming, the conventional freetype-based approach ought to suffice.
Such as: https://github.com/rougier/freetype-gl

c42f, 2016-10-15:
Thanks for that link, it seems that freetype-gl is not ftgl (a much older library), and does in fact support the signed distance formulation.  Unclear whether it'll work on windows and OSX though.

The annoying thing about library choice here is that I really don't want freetype and potentially a  bunch of other deps (harfbuzz?) when I've already got Qt for supposedly handling that stuff.  But as usual, if you don't want to do things "the qt way", then you're mostly out of luck :-(  So I can take on some extra deps, write a bunch of code, or try to shoehorn qt into doing what I want.

nigels-com, 2016-10-15:
Yeah, I never considered Qt first-class for OpenGL stuff.  _Give me a context and get out of my way_
But good enough nowadays?  Perhaps.

mcjohnalds, 2016-11-27:
QPainter can draw text straight onto a QImage. If that works out well enough I'll have a patch ready soon.

mcjohnalds, 2016-11-30:
Should the text shrink when you're further away or stay at a fixed size regardless of distance?

c42f, 2016-11-30:
Both of these seem like valid options in different circumstances.  To choose only one of them, I'd probably go with a fixed size for now, treating the text as data labels.

To generalize a bit, perhaps we could have a configurable minimum and maximum distance: closer than the minimum distance, the text has a fixed maximum size.  Further than the maximum distance it shrinks via the usual geometric scaling.  In between, it's a fixed size.

I'm not sure this makes sense, testing it out in a prototype would help.

c42f, 2017-02-24:
Closed by #160

closed 2017-02-24 by c42f