c42f/displaz #152: IPC : client not getting response

glandais, 2016-11-10:
Hi,
On a Windows 7 64bits, -querypoint and -hook are not always working on my computer.
When I run "displaz -querycursor", the message is sent to the GUI but response looks blocked somewhere. When the client times out, PointViewerMainWindow::handleMessage is executed.

In main.cpp, if I print on stdout before and after sending the message, it gets the response...
I tried a milliSleep but it didn't worked...

I'm available for any idea/bugfix!

c42f, 2016-11-11:
Ah, more IPC woes! It's surprisingly hard to make this work nicely on all platforms.  I've had problems with using qt sockets in synchronous mode before.  I worked around a related issue in 8364c38 (see also https://bugreports.qt.io/browse/QTBUG-47683, which has *still* not been addressed...), so the commentary there might supply a place to start looking.

In fact, looking through the code, I suspect the problem may be in `IpcChannel::receiveMessage()`, which calls `waitForReadRead()`, which IIRC is one of the *synchronous* `QLocalSocket` methods, and is probably broken on windows for the same reason that `waitForDisconnected()` is broken. I suggest that you have a read of the commentary in `IpcChannel::waitForDisconnected` and see whether you can run a similar local event loop as a workaround.

A further trawl through the code shows another place which a `waitFor` style synchronous method is called in `IpcChannel::connectToServer()` - this could also be a source of trouble given the general brokenness of this stuff on windows.

As a more general fix for the unreliability of the sychronous API, perhaps we could consider refactoring `IpcChannel` so that it always runs an event loop internally and uses the async functionality under the hood to simulate synchronous behaviour.

c42f, 2016-11-11:
If you're feeling particularly energetic, another IPC related problem is `InterProcessLock_test` (run with `ctest -R InterProcessLock_test`) which seems broken on windows for reasons I haven't had time to investigate.  I doubt this is the source of your current problem, however.

glandais, 2016-11-14:
It would have been great to refactor the event loop but it is a bit too hard for me...
Sadly I don't have enough space disk for Visual Studio on my C:, so I have to use Appveyor to build displaz... Then I'm also not able to fix the InterProcessLock_test...

glandais, 2016-11-14:
I reverted the modifications on `IpcChannel::connectToServer()`, as it locks the program exit when I'm using `displaz -script`...

cbecker, 2017-03-17:
Could this have something to do with `displaz -hook key:p cursor` not showing the key events on Windows? On linux it works perfectly, but on Windows it's not sending the key press messages back (I am using the event loop from Julia)

c42f, 2017-05-24:
Should be fixed by #153.

@cbecker yes this was definitely the problem with `-hook`

closed 2017-05-24 by c42f

cbecker, 2017-05-24:
Thanks, that's great news!