c42f/displaz #158: Built fails on a fresh install of Ubuntu 16.10

mcjohnalds, 2016-11-29:
```
cd build
cmake ..
make -j4
```

Fails with
```
...
[ 91%] Built target displaz-gui
CMakeFiles/unit_tests.dir/typespec.cpp.o:(.qtversion[qt_version_tag]+0x0): undefined reference to `qt_version_tag'
collect2: error: ld returned 1 exit status
...
```

nigels-com, 2016-11-29:
What's the version of cmake, and Qt?

c42f, 2016-11-29:
I've observed this in a couple of other machines now. In those cases it appeared when a version of conda shipped qt binaries into the LD_LIBRARY_PATH which were different from the ones which cmake found the headers for.

In that case, unloading conda from the environment fixed that.  Could this be something similar?

mcjohnalds, 2016-11-29:
I can't find anything related to conda on my system, nor have I ever installed it. It's a fresh install of 16.10 with only git, google-chrome, and the atom code editor installed.

As in my patch adding -DQT_NO_VERSION_TAGGING fixes things.

nigels-com, 2016-11-30:
Thanks for the information.  Seems like as good excuse as any to spin up a Yakkety docker container, to troubleshoot further.

c42f, 2016-11-30:
Exactly my thoughts, which is in fact what I just did.

The underlying problem seems to be that some Qt headers are coming in via an include of glutil.h into typespec.cpp, and the very action of transitively including qversiontagging.h causes a linker error, regardless of even calling any Qt functions.

Two obvious options present themselves: (1) just link `util_tests` against Qt::Core, even though it shouldn't be required (2) Disentangle typespec.cpp from the Qt headers entirely.

nigels-com, 2016-12-05:
For what it's worth, I don't see the harm in linking `util_test` against Qt::Core.

c42f, 2016-12-06:
Fixed in #161

closed 2016-12-06 by c42f