c42f/displaz #164: How to draw a box around a building (for example...)

mrosen, 2017-03-23:
First, let me say how impressed and thankful I am for this project.  A Cross-platform Hackable Lidar Viewer is dearly needed and this on has a very strong start.

What I'd like to do is be able to draw either a 2D or 3D polygon (or perhaps just a box) manually and get the corresponding points.  The intent is to load a point cloud, manually select some feature (manually recognized by a human) and extract the boundary.  What is a good way forward for this?

My modest and possibly misdirected first attempt was to manually query the 3D cursor to find the edge points, manually write a .ply file that outlines those points and manually load the file into the running displaz server.

c42f, 2017-03-25:
Hi Michael, thanks for the kind words, it's great you're finding the project useful!

Given that you're willing to write code, I think you're going about this the right way (or at least the way I envisage people using the tool): let displaz take care of the 3D view, but build the logic for your application inside a simple program, in whichever programming language makes you happy :-)

What I've been trying to do is expose UI events via the command line, which allows an event driven script, driven by keybindings in the displaz window.  If you look in the test directory, you'll see a script which does almost exactly what you want: https://github.com/c42f/displaz/blob/master/test/testhook_draw_polygons.sh, including incremental feedback. It's a `bash` script, so not exactly pretty, but hopefully you can see the idea, and see that it can be done in your language of choice.

Which language were you using to create your ply file, and which platform are you on, by the way? I haven't yet updated the simple language bindings (C,C++, matlab and python) to handle user interface callbacks. The julia bindings are more up to date in this regard (see https://github.com/c42f/Displaz.jl) and offer an event loop that you can hook into.

c42f, 2017-03-25:
By the way, it's sometimes hard to know which pieces to build into displaz, and which pieces to let people write themselves in a script. When you find things which are awkward or impossible from a script, I'm quite happy to entertain new features in the C++ code.

mrosen, 2017-03-28:
I haven't responded to this but I did find this hugely helpful.  I'm
working on a python-based script that will do what I need.  I'll be pleased
to contribute this when I'm done.  Displaz is a great starting point but
many things about it are not as discoverable as one might like.  Examples
will help.

msr

c42f, 2017-03-29:
Great!  As you say, the examples are lacking at this stage.  I did a big documentation update a while back, but the IPC system has moved on a lot since then.

c42f, 2017-04-07:
Closing this issue in favour of #166 - the underlying problem is documentation for the new event loop stuff, I suppose.

closed 2017-04-07 by c42f