c42f/displaz #173: visibility of vertices in combination with edges

Rheinwalt, 2017-06-27:
To begin with, great job! We use displaz for topographic data and are very happy with it. However, now I tried to visualize rivers together with the elevation data and failed. If I do the following I get only the edges and no vertices:

ply
format ascii 1.0
element vertex 14039154
property float x
property float y
property float z
property float r
property float g
property float b
element edge 1
property list uchar int vertex_index
end_header
+8.2237e+02 +9.9663e+01 +3.1556e+02 +2.0306e-01 +3.7972e-01 +5.5393e-01
+8.6825e+02 +9.3691e+02 +3.1270e+02 +1.6202e-01 +6.8732e-01 +4.9913e-01
...
9 0 1 2 3 4 5 6 7 8

Is it possible to visualize both, vertices and edges? The coloring of edges is fine by the way ..

In addition there seems to be a maximum edge list length, right? So more convenient would be something like mentioned at [paulbourke.net](http://paulbourke.net/dataformats/ply/):

element edge 5                        { five edges in object }
property int vertex1                  { index to first vertex of edge }
property int vertex2                  { index to second vertex }

Thank you!

c42f, 2017-06-30:
Glad you find it useful! I'm always interested in seeing a screenshot or two, if you have some nice data to show off ;-)

At the moment, displaz assumes that when there's "topology" present in the ply file (ie, connectivity between vertices  in the form of edges or faces) you don't want to show the vertices individually. This is a reasonable assumption in most cases. If you want both, the only way to get it right now is to just display two separate data sets, one for vertices and one for edges.  This is actually quite practical, so I suggest you try it and see if it works well enough.

For the other issue, if you use

```
property list uchar int vertex_index
```

You'll find that you can only have up to 255 edges in a list (the maximum size of a `uchar`).  But you can have many more, by just changing the `uchar` to `int`.

Rheinwalt, 2017-07-10:
Thank you for your tips! I will try this out. Having links and edges in two different datasets is actually quite convenient in this case. I'll post a screenshot of this when I'm done, until then:
http://imgur.com/a/zUrcH

Thanks!

c42f, 2017-07-11:
Neat.  Interesting color scheme - is that based on some kind of curvature measure for the point cloud?

Rheinwalt, 2017-07-12:
I know what you mean, but it's a network measure based on the flow in the landscape. The other one is a point density estimate. Crossing flight lines, vegetation, etc.