c42f/displaz #186: Installation on OS X/macOS

tillhainbach, 2018-05-08:
Hi,
I spent the last hours trying to get displaz workin on macOS 10.13.4. I somehow managed to get it running. But there are some issues, I'd like to share in case someone else wants to test displaz on a Mac.
1. one can basically follow the linux compile+install instructions but when run `displaz` in terminal it encounters the following error:
`dyld library not loaded @rpath/libQt5OpenGL.5.dylib`
` Referenced from: /usr/local/bin/displaz`
` Reason: image not found` 
as a workaround I copied the displaz file from the build folder to usr/local/bin
`cp ~/displaz/build/bin/displaz /usr/local/bin`
and it worked.
2. When opening a file I get weird output like
`objc[20178]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fff8a030c90) and /System/Library/PrivateFrameworks/FileProvider.framework/OverrideBundles/FinderSyncCollaborationFileProviderOverride.bundle/Contents/MacOS/FinderSyncCollaborationFileProviderOverride (0x12efd6cd8). One of the two will be used. Which one is undefined.`
This causes the displaz applications not being able to track the cursor position and clicks properly in the Finder window (at least that's my best guess). 
3. the shader parameters are not adjustable by clicking the little up/down arrows but somehow you can scroll them. Entering numbers works fine.

c42f, 2018-07-16:
Thanks for the feedback, I'm sure this will help some other mac users. It would be great to have a proper OSX installer but I can't do it myself without buying some OSX hardware for testing.

Alas, Apple have just deprecated OpenGL entirely which doesn't bode well for current displaz versions continuing to work on OSX...  A "serious" OSX package which is at least slightly future proof would probably need MoltenGL or some equivalent.

tillhainbach, 2018-08-21:
Hey,
I forked this displaz onto my repository and changed some stuff so that it builds a stable standalone Mac OS X Bundle. Unfortunately I haven't add the time to test if it still builds on Linux and Windows. There are however still some issues. The GUI is rendered strangely so that buttons, etc have the charm of Windows 98. I also add to trick the bundle into behaving like a Mac OS X bundle (by double-clicking displaz.app it does not launch displaz executable but it launches a launcher bash script that in turn launches the actual displaz executable). Without this workaround displaz.app launches two processes: 1. a displaz process that keeps jumping up-and-down in the menu bar
2. a Display process that actual displays the GUI and works fine

I tried to google a little bit to find the reason for that behavior but I still don't have the slightest clue.

I mostly changed the CMakeFiles and rearranged the directory. The only thing I had to change in the source code was in guimain.cpp. 
There is a line where the directory to the shaders is created. Originally, it was done by .chop(4) to cut of /bin from eg. usr/local. In a Mac Bundle, the executable resides in displaz.app/Contents/MacOS so I had to find a way to chop of both /MacOS or /bin. 

QString installBinDir = QCoreApplication::applicationDirPath();
QString installBaseDir = installBinDir.mid(0, installBinDir.indexOf("/",-7));

evetion, 2019-10-23:
Just compiled displaz succesfully on MacOS Catalina. I used macports to install qt5 and apart from the thirdparty libs generating makefiles in their own folder instead of the folder I was in it works.

Note that port will take ages compiling qt5, but the brew version didn't work. Downloading the official qt5 framework would also be an option, but I didn't test that.

> one can basically follow the linux compile+install instructions but when run displaz in terminal it encounters the following error:
dyld library not loaded @rpath/libQt5OpenGL.5.dylib
Referenced from: /usr/local/bin/displaz
Reason: image not found
as a workaround I copied the displaz file from the build folder to usr/local/bin
cp ~/displaz/build/bin/displaz /usr/local/bin
and it worked.

I used `make install` which installs to `/usr/local/bin/displaz`.

> objc[20178]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fff8a030c90) and [...]

Can't reproduce this one.

> the shader parameters are not adjustable by clicking the little up/down arrows but somehow you can scroll them. Entering numbers works fine.

This is probably by design, the up down arrows indicate a scrolling enabled field? Can confirm otherwise.

c42f, 2019-10-24:
The up down arrows thing is honestly a hack. The idea is to click, hold and *drag* the mouse giving you analog control rather than in single steps. No idea if it works on MacOS though, I did hook this behaviour into the system in a weird way.

evetion, 2020-11-21:
I've tried this again on MacOS Big Sur and Qt5.15 but no luck thus far. I can get it to build, but it won't display points, errors with:
```
Geometry :: vertexArrayObject was not found - points
Geometry :: vertexBufferObject was not found - point_buffer
```
Any tips to dive into the root cause here?

c42f, 2020-11-23:
My suggestion for debugging any tricky OpenGL problem is to run `displaz` with apitrace: https://apitrace.github.io/

evetion, 2020-12-16:
On macOS Big Sur it works again 😄 , after the Qt5 port files were updated. I've used `cmake .. -DCMAKE_PREFIX_PATH=/opt/local/libexec/qt5/lib/cmake`. I had to do a `make install`, the `bin/displaz`executable didn't work, something something paths I guess.

evetion, 2022-01-11:
I can confirm that compiling works fine on a M1 Mac. Just use brew to install the required tools, set PATH to include the qt5 folder and it runs. 🎉