c42f/displaz #188: Re-rendering whenever move the inside the viewer

quhezheng, 2018-08-08:
Load a las file up to 2.0G bytes, it's very slow to render, but more points are coming up in the viewer little by little. 
When I move the mouse, the viewer restart the rendering from scratch. This is very painful, anyway to fix it?

nigels-com, 2018-08-08:
I'd suggest splitting the point cloud into seperate (smaller) files.

nigels-com, 2018-08-08:
It so happens we have a (commerical) solution for that.
But that's also based on processing large point clouds (multi TiB) into manageably small pieces.
See: https://www.youtube.com/watch?v=XvYcvcGiAj0

c42f, 2019-03-28:
Viewing large las files should absolutely be possible (this was the main design goal for displaz!) and 2GB should not be a problem.

It sounds like either (a) you have a very slow graphics card, or (b) some kind of bug in displaz's incremental rendering system.

In either case you'd have to provide more information to allow the problem to be tracked down. How many points do you have? From memory I was able to render on the order of 100 million points with displaz last time I was using it heavily.