c42f/displaz #191: Problem with build.

IakovK, 2018-11-30:
I tried to build the displaz against Visual studio 2017 and qt 5.11.1 - according to the readme. Build was successful, but run was not:
When running, I see several message boxes like this:
---------------------------
displaz.exe - Точка входа не найдена
---------------------------
Точка входа в процедуру ?textureListFlags@QWidgetPrivate@@UEAA?AV?$QFlags@W4Flag@QPlatformTextureList@@@@XZ не найдена в библиотеке DLL D:\Projects\Yurevich\software\displaz\build\bin\Release\Qt5OpenGL.dll. 
---------------------------
ОК   
---------------------------
(Entry point not found)
The CMake generator was Visual Studio 15 Win64.

c42f, 2018-12-12:
I've never seen anything exactly like this. But at a guess it looks like the version of the Qt DLLs which are found nat runtime are not the same as the ones you used when building the program?

Was this when running from the build directory? If so, you should probably install displaz first before running. Another thing to try is to create the installer package, run that to install displaz and then run the installed version.

c42f, 2018-12-12:
Closing, as this is probably an issue with the local build config rather than with displaz itself. Feel free to provide some more information though, and we can reopen if necessary.

closed 2018-12-12 by c42f