c42f/displaz #192: Build failure on Windows Qt 5.11 VS2015 x64

TheManojPal, 2019-10-11:
The build fails on windows with Qt5.11 and visual studio 2015 x64. 

Errors at 
QRect geom = QApplication::desktop()->screenGeometry(this); 

error C2039: 'desktop': is not a member of 'QApplication'
note: see declaration of 'QApplication'

error C3861: 'desktop': identifier not found
error C2227: left of '->screenGeometry' must point to class/struct/union/generic type
note: type is 'unknown-type'

I am new here. Please advise.

c42f, 2019-10-12:
Looks like `desktop` is deprecated and to fix this on recent Qt versions it would need to be replaced with something equivalent. Will require some investigation given that both uses of `desktop()` are basically hacks.

Not sure when I'll get time to look into this. A simple workaround would be to compile with an older Qt version. It looks like version 5.9 still works.

nigels-com, 2023-01-21:
The Ubuntu 18.04 (bionic) Qt version is 5.9.5.
For Ubuntu 20.04 (focal) Qt version is 5.12.8.
For Ubuntu 22.04 (jammy) Qt version is 5.15.3.
Ideally we'd try to support 5.9 onwards, IMHO.

nigels-com, 2023-01-21:
Built master just now on Windows with Qt 5.15.2

nigels-com, 2024-01-28:
C++17 is now required.  It's supported by VS2019 or newer.

closed 2024-05-01 by nigels-com