c42f/displaz #205: Error loading .PLY files under Manjaro

max-hess, 2022-04-12:
Hi,

I'm running Manjaro Gnome 21.2.5 and built displaz according to the readme.rst.

I get an error when I open *.PLY files which include floats. Log message says:

ERROR: rply: Error reading 'x' of 'vertex' number 0.
ERROR: Could not load ... (path/file.PLY)

[image]

The error doesn't occur if the vertices are written as integers. Also, other files such as *.las work completely fine.
The test file which doesn't open looks like this:

ply
format ascii 1.0
element vertex 4
property float x
property float y
property float z
end_header
1.5 1.5 1.5
1.5 1.5 2.5
1.5 2.5 2.5
2.5 2.5 1.5

The version with integers:

ply
format ascii 1.0
element vertex 4
property float x
property float y
property float z
end_header
1 1 1
1 1 2
1 2 2
2 2 1

Here are the terminal outputs for the 2 make -j4 commands :
[cmake_output.txt](https://github.com/c42f/displaz/files/8472239/cmake_output.txt)
[cmake_output_2.txt](https://github.com/c42f/displaz/files/8472240/cmake_output_2.txt)

Not sure where the problem comes from.

Rheinwalt, 2022-04-12:
I copy pasted your examples into plain text files and they opened just fine. Looking at your screenshot and reading the rply error message makes me wonder whether your system interprets floating point numbers only as such if the point is a comma. Like it is for instance common in Germany. I got that idea because the floating point shader parameters are also displayed with commas instead of with a point like it is done for me. Could you test your example file with commas instead of points? Did you set your LC_NUMERIC or something?

max-hess, 2022-04-12:
You are right! 
Works fine after changing the LC_NUMERIC to something like en_US.UTF-8.
Thanks!

closed 2022-04-12 by max-hess