c42f/displaz #229: not compatible with lastools 2.3 or recent pdal

landryb, 2024-04-22:
while building against external lastools 2.3.0, the build fails with
```
/usr/obj/ports/displaz-0.4.0/displaz-0.4.0/src/las_io.cpp:185:49: error: no matching constructor for initialization of 'LASreaderLAS'                                                 
    std::unique_ptr<LASreaderLAS> lasReader(new LASreaderLAS());                                                                                                                      
                                                ^                                                                                                                                     
/usr/local/include/LASlib/lasreader_las.hpp:79:3: note: candidate constructor not viable: requires single argument 'opener', but no arguments were provided                           
  LASreaderLAS(LASreadOpener* opener);                                                                                                                                                
  ^                                                                                                                                                                                   
/usr/local/include/LASlib/lasreader_las.hpp:61:18: note: candidate constructor (the implicit copy constructor) not viable: requires 1 argument, but 0 were provided                   
class LASLIB_DLL LASreaderLAS : public LASreader     
```

i've tried building against pdal instead of lastools, but with version 2.7, and the current code uses apis that were removed in PDAL 1.2.
```
/usr/obj/ports/displaz-0.4.0/displaz-0.4.0/src/las_io.cpp:83:45: error: no type named 'Reader' in namespace 'pdal'
    pdal::Stage* reader = static_cast<pdal::Reader*>(manager->getStage());
                                      ~~~~~~^
/usr/obj/ports/displaz-0.4.0/displaz-0.4.0/src/las_io.cpp:90:11: error: no type named 'PointBufferSet' in namespace 'pdal'
    pdal::PointBufferSet pbSet = manager->buffers();
    ~~~~~~^
/usr/obj/ports/displaz-0.4.0/displaz-0.4.0/src/las_io.cpp:90:43: error: no member named 'buffers' in 'pdal::PipelineManager'
    pdal::PointBufferSet pbSet = manager->buffers();
                                 ~~~~~~~~~^
/usr/obj/ports/displaz-0.4.0/displaz-0.4.0/src/las_io.cpp:92:24: error: use of undeclared identifier 'context'; did you mean 'connect'?
    pdal::PointContext context = manager->context();
```

the pipeline reader has been merged into pipeline manager, cf https://github.com/PDAL/PDAL/blob/282386ee59e4c204d6400d5813f2a0be82f29312/RELEASENOTES.txt#L1579

landryb, 2024-04-22:
@nigels-com since you seem to be the new maintainer, any idea about that ?

landryb, 2024-04-22:
@hobu since you contributed the pdal support in 1e2e688283255df50342af5779abeab0ed473ef6 , your hindsight will be welcome :)

nigels-com, 2024-04-22:
@landryb Thanks for the issue, time to migrate, I guess!

nigels-com, 2024-04-24:
@landryb The current LASTools release is [v2.0.3](https://github.com/LAStools/LAStools/releases/tag/v2.0.3).  Is 2.3 a typo?

nigels-com, 2024-04-24:
Never mind.  Having the same issue with v2.0.3.

nigels-com, 2024-04-24:
Try: PR #231

landryb, 2024-04-25:
yes, it builds fine with #231 !

nigels-com, 2024-04-25:
Should work fine, I have not done any testing yet though!

nigels-com, 2024-04-28:
Fixed.

closed 2024-04-28 by nigels-com