c42f/displaz #247: feature request: add .shp or .kml file to the model for a very good experience

luiscuadrado1234, 2024-07-15:
Hello everyone.

Diplaz is a very good program, simple, lightweight and free, to be able to deliver work as a viewer to clients. But many times to make it a better experience, the option to add external files to the point cloud visualization would be missing, usually vector formats such as lines or polygons.

I suggest, as an example, formats like shapefile or KMZ/KML

nigels-com, 2024-08-18:
I'd like vector files such as WKT and KMZ too.
But one issue is where to place the 2D ones, if we don't know ground level.

luiscuadrado1234, 2024-09-01:
Hello.
I think that, on a first impression, it could be placed as a "painted line" above the point cloud, that is, not at ground level... like the "clamp to the ground" style of the KMZ files